jamwaffles/optiplower #6: DS2502 sense

jensvanhoof86, 2024-03-04:
Hi,

I have just programmed a DS2502 with the following code "DELL00AC090195046CN09T" using Gary Stofer's programmer sketch. I have had no luck with the Attiny chips so far so I figured I'd give the actual DS2502 a shot.

I have the DS2502 currently on a breadboard with GND connected from the optiplower to the chip, and the datapin of my 'dell' barrel connector connected to the data pin of the DS2502. Unfortunately, no luck.

Looking with a logic analyzer, I see the following with an original Dell adapter:

-as long as the barrel connector is not in the computer, sense line is low
-when the connector goes in the computer, the sense line is pulled high and the data communication is initiated next. After the data exchange, the sense line idles back high.

However, with my setup, the sense line is noisy even before I plug the barrel connector in so the data is obviously not exchanged correctly and it doesn't work.

Any ideas what I could be missing? Another external pull-up/down resistor required?

jamwaffles, 2024-03-04:
I'm afraid it's been a while since I worked on this project last so my memory is fuzzy, but I think you might need a ~10k pullup. I didn't write any of the code in this repo, just hacked it up for ATTiny, but you might want to look through it to see if there is an internal pull up/down enabled anywhere.

I also vaguely remember having to connect the NC (Not Connected :roll_eyes:) pin on the DS2502 to get it to work, but I could be wrong there...

jensvanhoof86, 2024-03-04:
In the programmer there's a 4.7k pullup, but in theory I'd expect that pull-up on the 'master' side, so within the PC.
I did see in your schematic that you have the NC pin connected to GND. I'll try to investigate further. Thanks!